TMRh20/TMRpcm #122: can the Recording.ino work to SPIFFS, without SD card on ESP32?

ligantx, 2020-04-08:
Recording.ino uses SD card ChipSelectPin to write the audio file
`audio.CSPin = SD_ChipSelectPin;`

Can this work with an adjustment and write the audio file to SPIFFS of ESP32?

TMRh20, 2020-04-08:
Nope, this library only supports AVR devices. The ESP internals are so much different, it would require a complete rewrite.

Funny you ask this now however, because I am just waiting on hardware to arrive and I will be adding ESP32 support to my AAAudio library. Well, as long as I can figure out how...

AAAudio allows a lot more than just recording and playback and is the successor to TMRpcm. It would allow SPIFFS recording on ESP32.

See https://github.com/TMRh20/AutoAnalogAudio

ligantx, 2020-04-08:
nice!
thanks for pointing out..
i guess if i change the code[ here](https://github.com/TMRh20/AutoAnalogAudio/blob/master/examples/SDAudio/SdAudioRecording/myRecording.h):

`recFile = SD.open(fileName,FILE_WRITE); `
to 
`recFile = SPIFFS.open(fileName,FILE_WRITE);`

its ready to go!

closed 2020-05-11 by TMRh20