TMRh20/TMRpcm #124: DISABLE_SPEAKER2

cebab54, 2020-05-20:
I am using a UNO R3 Board and this does not seem to do anything.  There is still a signal on Pin 10 as well as Pin 9.  I had assumed that Pin 10 would become 'silent' and unused when this OPTION was enabled?

TMRh20, 2020-05-20:
Hmm, as long as you haven't enabled `#define USE_TIMER2` , it will prevent the timer output on pin10 from being enabled.

To verify, add the following code in your setup() section:
  ```
Serial.print("Test ");
Serial.println(TCCR1A,BIN);
```
With Speaker2 enabled: Test 10110010
With Speaker2 disabled: Test 10000010

If you get different results, try removing TMRpcm using Arduino library manager, then open your Arduino library folder (Documents->Arduino->Libraries) and delete any remaining TMRpcm folders. Then reinstall using the IDE.

TMRh20, 2020-05-20:
Should probably add, you want to attempt to play an audio file before calling the above test code.

cebab54, 2020-05-21:
Hi

Tried your test as suggested.  Result is same in both cases: With Speaker2 enabled or disabled:

Test result = 10110010 (in both states)

Deleted TMRpcn library files and re-installed using IDE (version 1.8.12) as suggested and result of test is exactly the same in both states.  So still not working.

Andrew

TMRh20, 2020-05-21:
Ok, I have one more idea.
  
In the IDE, if you enable File->Preferences->Show Verbose Output During: Compilation , it should show something like this after compiling:  
` Using library TMRpcm at version 1.1 in folder: C:\Users\tmrh20\Documents\Arduino\libraries\TMRpcm `
  
Then of course just verify that you are editing the pcmConfig.h file in the folder where it is finding the TMRpcm library.

You can also call the following in your sketch to see if the define is being picked up at all:
```
  #if defined (DISABLE_SPEAKER2)
    Serial.println("SP2 Disabled");
  #endif
```

cebab54, 2020-05-21:
Hi

Thanks.  The Define statements are working after adding the lines:

  #if defined (DISABLE_SPEAKER2)
    Serial.println("SP2 is Disabled");
  #endif

And not when commented out.

However, these statements do stop the audio.play() call to stop working.

When removed the clip plays correctly.

Another thing I have noticed is that the audio.play(“clipname.wav”,1) does not work. And the clip is not played at all, whereas using audio.play(“clipname.wav”) works correctly.  Is this a limitation of using the UNO board?

Andrew S Hudson

8 Kingswood Avenue,
Bromley BR2 0NY
m: +44 (0) 7836 349436

cebab54, 2020-05-21:
Meant to say, although the define seems to be working as per tests and output to Serial Monitor, the Test output lines no longer work to show the Binary output AND there is still signal coming from Pin 10 even when the SPEAKER2 is disabled.

Andrew S Hudson

cebab54, 2020-05-21:
Hi

There is something really weird going on.  I have uninstalled and reinstalled the TMRpcm Library now 4 times and every second time I run the IDE Compiler, there seems to be something overwritten that is stopping your Library from working correctly after that compile.

That said this time the compiler does not seem to be stopping the Library from working, but still there is signal on SPEAKER2, pin 10 despite SPEAKER2 test indicating it is OFF i.e.
20:49:56.790 -> Test 10000010
It is just a bit frustrating …. As I am sure you can appreciate …

Andrew S Hudson

TMRh20, 2020-05-21:
Are you using the included examples or custom code? Any other libraries being used? If you are low on memory or there is a related issue like a buffer being overwritten etc. that is a common source of "weird" issues.

> audio.play(“clipname.wav”,1) does not work... Is this a limitation of using the UNO board?
Nope, the code is general to all boards.

cebab54, 2020-05-22:
Hi

The code uses parts of the example code.  The RAM free is about 300

The Sketch is as follows:

#include <SD.h>
#include <TMRpcm.h>
#include <SPI.h>
#include <pcmConfig.h>
#include <BirdData.h>
#define SD_ChipSelectPin 4
#define LDR_ChipSelectPin 0
#define DIGI_LineoutSelectPin 9
#define AUDIO_Volume 4
#define EXT ".wav"
Sd2Card card;
TMRpcm audio;

const unsigned int DATA_SIZE = 11;
const boolean DEBUG = false;
BirdData data[DATA_SIZE];

// helper function to determine free ram at runtime
int freeRam () {
  extern int __heap_start, *__brkval;
  int v;
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
}

void(* resetFunc) (void) = 0; //declare reset function at address 0

void setup() {

  audio.speakerPin = DIGI_LineoutSelectPin;
  audio.setVolume(AUDIO_Volume);  // reset output volume
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  //  #if defined (DISABLE_SPEAKER2)
  //    Serial.println("SP2 is Disabled");
  //  #endif
  //  #if defined (HANDLE_TAGS)
  //    Serial.println("HANDLE_TAGS is enabled");
  //  #endif

  Serial.print("Free Ram: ");
  Serial.println(freeRam()); Serial.flush();

  Serial.println("Initializing SD card...");
  // if (!card.init(SPI_HALF_SPEED, SD_ChipSelectPin)) {
  // check if an SD card is present and can be initialized:
  if (!SD.begin(SD_ChipSelectPin)) {
    Serial.println("SD Card initialization failed:");
    Serial.println("* is a card inserted?");
    Serial.println("* is your wiring correct?");
    Serial.println("* did you change the chipSelect pin to match your shield or module?");
    while (1);
  } else {
    Serial.print("Card type found is: ");
    switch (card.type()) {
      case SD_CARD_TYPE_SD1:
        Serial.println("SD1");
        break;
      case SD_CARD_TYPE_SD2:
        Serial.println("SD2");
        break;
      case SD_CARD_TYPE_SDHC:
[... 180 lines cut ...]
    }
    delay(100);
    audio.loop(0);  // set play loop function off
    Serial.println("Now finished current clip output.");
    Serial.flush();
    delay(100);
  } else {
    Serial.println("Night mode: No sounds to loop.");
    Serial.flush();
    delay(100);
  }
  return;
}

The only thing I get is a warning on compile that states as follows:

In file included from C:\Users\ashud\Documents\Arduino\libraries\arduino-bird-scarer\arduino-bird-scarer.ino:2:0:

C:\Users\ash\Documents\Arduino\libraries\TMRpcm/TMRpcm.h:31:7: note:   initializing argument 1 of 'void TMRpcm::play(char*)'

  void play(char* filename);

Thanks for your interest

Andrew S Hudson

TMRh20, 2020-05-24:
As I said, the problem is going to be with your code. The library tests OK.

cebab54, 2020-07-11:
Hi again

It has been a while since you sent the email below.  Many thanks for that feedback. The COVID Lockdown has become rather tedious until recently and I was persuaded to spend time decorating  …. Which kept me busy instead of focussing on the Arduino project. But my attention has turned back to the Arduino now.

One quick question.  Can the sound files for TMRpcm have ASCII based string names like “sound1.wav” or should they simply be named as “1.wav”?

Thanks

Andrew

Andrew S Hudson

TMRh20, 2021-10-30:
Sound files can have ASCII strings names.

The stock SD library uses the Sd2Card class

closed 2021-10-30 by TMRh20