TMRh20/TMRpcm #125: TMRpcm Slow Audio Play Back

HobbyFalcon, 2020-06-13:
I have hooked up a micro sd card to the Arduino pro mini.  I am using the TMRpcm library to play the wav files that I have converted to 8 bits 32khz mono, as instructed.  I can get the audio to play but the speech is slowed down by at least a factor of 2 or more.  Any ideas on what I've done wrong?

Heres the code I am using:
`#include <SD.h>                      // need to include the SD library
//#define SD_ChipSelectPin 53  //example uses hardware SS pin 53 on Mega2560
#define SD_ChipSelectPin 10  //using digital pin 4 on arduino nano 328, can use other pins
#include <TMRpcm.h>           //  also need to include this library...
#include <SPI.h>

TMRpcm tmrpcm;   // create an object for use in this sketch

void setup(){

  tmrpcm.speakerPin = 9; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc

  Serial.begin(9600);
  if (!SD.begin(SD_ChipSelectPin)) {  // see if the card is present and can be initialized:
    Serial.println("SD fail"); 
    return;   // don't do anything more if not
  }
  tmrpcm.quality(1);
  tmrpcm.volume(2);
  tmrpcm.play("1.wav"); //the sound file "music" will play each time the arduino powers up, or is reset
}

void loop(){ 

  if(Serial.available()){ 
    if(Serial.read() == 'p'){ //send the letter p over the serial monitor to start playback
      tmrpcm.play("1.wav");
    }
  }

}
`

TMRh20, 2020-06-13:
The library will typically auto-detect the sample rate etc, so if I remember correctly this would usually be due to the file actually being in stereo or 16-bit format.  
Probably a good idea to verify that your pcmConfig.h is set to the defaults per https://github.com/TMRh20/TMRpcm/blob/master/pcmConfig.h as well
Have you tested with the included example files?

HobbyFalcon, 2020-06-13:
Yes they all have the same result of being twice as slow.  Also I verified that my audio is 8 bit and mono, so that is not the problem.  It just occurred to me however that I have been running this on a 3.3 v, 8 MHz board.  Does this library require a 16 MHz processor to run properly?

TMRh20, 2020-06-13:
Yeah it expects F_CPU to be defined. You can probably just define it ( #define F_CPU 8000000 ) at the beginning of TMRpcm.cpp

HobbyFalcon, 2020-06-13:
I tried that but I got the same result. I am going to try it with a 16 MHz board in the morning.  Thank you for your help.
Also I am not sure if I updated the TMRpcm.cpp correctly.  I went into arduino/libraries and opened it then changed the code, but is there a way that I'm supposed to re-download it to ensure the change is made in IDE?

HobbyFalcon, 2020-06-13:
Okay I ran it on a 16 MHz board with a voltage divider to get it down to 3.3 v and it worked.  Thank you for your help!

closed 2020-06-13 by HobbyFalcon