TMRh20/TMRpcm #126: Offset of stereo audio with 2 output pins

Sioreto, 2020-06-23:
Hi,
I am currently doing a project for my university in which I need to obtain a stereo audio output from an Arduino UNo/Mega. I used the advanced function of the library TMRpcm and I defined STEREO_OR_16BIT and I used all the code necessary. However, for the project I need to know how much is the DC offset coming out of the pins. Can you help me please?

TMRh20, 2020-06-23:
Hmm, not sure what you are looking for exactly, but the Arduino outputs are 0-5v, so the general "DC offset" would be around 2.5v.  
  
If you are looking for the DC offset of an audio signal being output by the Arduino, I believe you would need an oscilloscope to measure the waveform and determine the DC offset.

Sioreto, 2020-06-23:
The second is what I was looking for. Actually, I thought that in the algorithm of the function audio.play() there was a preset offset. Isn't there? Thank you.

TMRh20, 2020-06-23:
The “volume” setting would affect the offset of the signal, but otherwise there is no preset offset. The wav/pcm data is generally fed to the timers as-is.

closed 2021-10-30 by TMRh20