TMRh20/TMRpcm #129: STEREO_OR_16BIT Problem

Sioreto, 2020-06-30:
Hi,

I am working with an Arduino UNO and I need to obtain a stereo audio output. When I tried to use a mono output the sound was very clear, but in stereo mode (uncommnenting STEREO_OR_16BIT and also bufferSize 128) in both outputs the tracks are heardable but the noise is too high in both outputs. I tried to reduce the sample rate of the track and also to vary setVolume() and quality() but the problem remains. I read that even if the CS is set to pin 4 in UNO, the pin 10 must be left unused. Is it because of that? Is it better if I use a MEGA?

Thank you

TMRh20, 2020-06-30:
>  I read that even if the CS is set to pin 4 in UNO, the pin 10 must be left unused. Is it because of that?

No

>Is it better if I use a MEGA?

Not really. More memory is available but a bigger buffer won't affect performance much.  

What kind of noise is it? Distortion, high pitched sound etc?
Have you tried multiple audio tracks to ensure its not a source issue?
Are the tracks properly encoded? ( 8-bit, 16khz, stereo is probably best to test with)
How are you connecting your output device in mono & stereo modes? What are you using?
Not really sure what would cause this.

Sioreto, 2020-06-30:
Thanks for the reply. There is an high pitched sound which overlaps the actual track. I tested both with stereo 8 bit 16kHz and 8 kHz. I am using headphones with a mono jack (both ears receive the same track)  and I test pin 9 and 10 separately. The headphone jack is low pass filtered with a resistor of 150 and a capacitor of 0.1 microF.

TMRh20, 2020-07-01:
Hmm, with an 8khz sample rate and quality(); set to 0, a high pitched sound would be expected because the 8khz pwm signal itself is audible. With quality set to 1, the timers run at 2x speed (16khz) so that should not generally be a problem.  
Have you tested the converted stereo track on PC or other device to ensure the audio is clean? Or tested with another audio track?
You could try a higher sample rate (24khz), but I doubt if that will make a difference.

closed 2021-10-30 by TMRh20