TMRh20/TMRpcm #131: Using stereo mode with Ethernet Shield

Sioreto, 2020-07-08:
Good morning,
I am trying to obtain a stereo output using STEREO_OR_16BIT with an Arduino Uno and an Ethernet shield. Unfortunately, when I use the normal SD card module I am able to get the correct output at both pin 9 and 10, while if I use the shield I only perceive a loud high-pitched sound at both pins. Is there a way to obtain stereo output and using the Ethernet shield?
Thank you

TMRh20, 2020-07-08:
Nope, the shield is hardwired to pin10 for SS/CS operations, so you would probably have to physically break that connection and modify the ethernet library to accommodate a different pin.

Sioreto, 2020-07-09:
Thank you! And would it be possible to get stereo audio with the Mega and the Ethernet shield?

TMRh20, 2021-10-30:
Should be possible

closed 2021-10-30 by TMRh20