TMRh20/TMRpcm #133: open txt file and play wav file

genia1991, 2020-08-16:
Hi everyone. I have an issue trying to open not only wav file. I wanna open txt file parallelly. Is there any how to do that ?
useing just SD library. Thanks for help

TMRh20, 2020-08-16:
Can't currently access the SD card while a file is playing, but if you use `tmrpcm.isPlaying();` to prevent opening while playing a file, or `tmrpcm.stopPlayback();` to stop playback before accessing SD, then it would work.

I'm not sure why this doesn't work, since the library can open and read multiple files for multi-track playback, but opening one from the sketch causes issues. I'll have to look into it further.

TMRh20, 2020-08-17:
In looking at this further I am fairly certain it has to do with the core functionality of the library.

AVR micro-controllers are typically single threaded, so even though a user may open 2 files from SD card, only one of them will be accessed at any given time.

TMRpcm uses interrupts to achieve asynchronous playback, and even when two files are open for playback, they are never accessed at the same time. 

When opening a file from the main sketch, if audio is playing, it will literally interrupt the code execution of the main loop, trying to read from file at the same time another file is open/being accessed. The SD lib is not designed for this and it won't work.

tl;dr: In short, this will never work during playback.

closed 2020-08-17 by TMRh20

TMRh20, 2020-08-17:
Just FYI I was reminded of my own library Auto Analog Audio, which would allow this because it can play audio without using interrupts. 

http://tmrh20.github.io/AAAudio/SDAudioBasic_8ino-example.html

genia1991, 2020-08-18:
I am very appreciate for your help and I find every your reply very useful. This helped me to solve the problem :)