TMRh20/TMRpcm #138: 16 bit mono

alittlebitweird, 2020-11-01:
Am I understanding the advanced section correctly? Is it possible to play back 16bit mono audio via a resistor ladder? Or is that specifically for running 8bit stereo?

Your library is amazing and perfect for what I want to do, wav file manipulation-wise, but the 8bit quality limitation is proving to be a problem for me.  I tried going with other boards using i2s, which sounds great, but none of the libraries ive found have the ability to play back a wav file at a specific seek point, like tmrpcm. Any advice?

Thanks for doing great work

TMRh20, 2020-11-01:
Yeah, in theory it’s 16-bit with ladder but quality wise your prob better with a 8-bit and higher sample rate. I should be adding esp32 support to my auto analog audio library by the end of the year.

TMRh20, 2020-11-06:
Beginnings of using I2s on the ESP32: https://tmrh20.blogspot.com/2020/11/esp32-playing-around-with-i2s.html
One could change the file starting point of the example sketch from 44 to a higher value to advance the starting point.

closed 2020-11-06 by TMRh20