TMRh20/TMRpcm #139: Sampling Rate Too High

austinddeshong97, 2020-11-04:
I'm trying to use this library with an Arduino Pro Mini and ATMega 328P, and TMRpcm is throwing
```
WAV ERROR
SAMPLE RATE TOO HIGH: 24000
```
I tried using .WAV files from another project, and ones I converted myself. For the ones I converted myself, I made them 8-bit, 16khz sample rate, mono, and PCM Unsigned 8-bit.

This is the code I'm running to try and get this working. Any help would be appreciated. Can't seem to find any answers anywhere.
```
#include <SD.h>                      // need to include the SD library
//#define SD_ChipSelectPin 53  //example uses hardware SS pin 53 on Mega2560
#define SD_ChipSelectPin 10  //using digital pin 4 on arduino nano 328, can use other pins
#include <TMRpcm.h>           //  also need to include this library...
#include <SPI.h>

TMRpcm tmrpcm;   // create an object for use in this sketch

void setup(){

  tmrpcm.speakerPin = 9; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc

  Serial.begin(9600);
  if (!SD.begin(SD_ChipSelectPin)) {  // see if the card is present and can be initialized:
    Serial.println("SD fail");  
    return;   // don't do anything more if not
  }
  tmrpcm.play("ON.wav"); //the sound file "music" will play each time the arduino powers up, or is reset
}

void loop(){  

  if(Serial.available()){    
    if(Serial.read() == 'p'){ //send the letter p over the serial monitor to start playback
      Serial.println("Got it!");
      tmrpcm.play("ON.wav");
    }
  }

}
```

TMRh20, 2020-11-04:
That error should only come up if sample rate is > 45050. I would verify that the file is encoded properly. I usually  suggest using Audacity.

austinddeshong97, 2020-11-04:
Just tried making a fresh file, and formatted my microSD as well and still had the same issue

TMRh20, 2020-11-04:
Same problem https://github.com/TMRh20/TMRpcm/blob/master/TMRpcm.cpp#L479 that nobody else is reporting.

closed 2020-11-04 by TMRh20