TMRh20/TMRpcm #143: Empty recording files

policarpobaquera, 2020-12-05:
Hello,

I am trying to test the audio recording capabilities of the library and I cannot run properly the _recording.ino_ example. The files that the program writes on the SD card are created but no sound recorded. I am using an Arduino Mega 2560 Pro and I am uncommenting the following settings in _pcmConfig.h_:

#define buffSize 254
#define SD_FULLSPEED
#define ENABLE_RECORDING
#define BLOCK_COUNT 10000UL

I have tested the library for playback on my speaker and it also seems to work correctly. 

Thanks for such a great library!

policarpobaquera, 2020-12-06:
Hello again,

I have realized that once the speakerPin is defined, the audio recording capabilities do not work anymore. I am able to play sounds and record independently but not both in the same Arduino routine. I am working on an audio recorder prototype. Is there any way to work around this issue?

TMRh20, 2020-12-07:
The recording functionality was never working well beyond proof of concept, the AVR devices just aren’t quite fast enough. I had it working with both recording and playback at the same time though, so not sure what the problem would be.

I would suggest using a more powerful device like Due or ESP32 and possibly looking at the AAAudio library http://tmrh20.github.io/AAAudio/

policarpobaquera, 2020-12-08:
Thanks for the help! After reassembling the prototype from scratch, I realized where the problem was: the speaker was connected to pin 46 in an Arduino Mega 2560 Pro Mini and if that pin was selected as a speaker ( `audio.speaker(46)` ), the recording feature didn't work at all. I finally made it work exactly as in the example, setting pin 11 as a speaker and 12 as an output (although I was not using it).

For me, this library has been working great, and upgrading my project from Nano to Mega 2560 Pro Mini made the prototype more than efficient to record and playback several audio tracks with decent quality. Having said that, I am eager to check your other AAAudio library with an ESP32.

Again, thanks for answering so quickly and developing this library!

closed 2020-12-08 by policarpobaquera