TMRh20/TMRpcm #145: esp8266 and esp32 compability

AloneFe, 2020-12-22:
Hey is there any way to use the recording future on the ESP8266 or ESP32?
and is there any way to modify the .wav file? make it less noise when recording?

TMRh20, 2020-12-23:
Esp32 is supported in the aaaudio library: https://github.com/TMRh20/AutoAnalogAudio

Using batteries or a cleaner power supply can reduce noise when recording.

Iftikhar-Omar, 2021-02-08:
> Esp32 is supported in the aaaudio library: https://github.com/TMRh20/AutoAnalogAudio Using batteries or a cleaner power supply can reduce noise when recording.
> […](#)

Please help with a more detailed guide. Most preferably step by step guide on how to use it with esp32.

Theagainmen, 2021-05-10:
Would like to see this too, any updates?

closed 2021-10-30 by TMRh20

mahmou2001, 2024-03-01:
can you help me  i can't use the lib to record with esp32 with max9814

TMRh20, 2024-03-01:
The ESP32 is not compatible with the TMRpcm library. I've tried to get it working with my other library, AutoAnalogAudio, but I don't believe the device is capable of recording without glitches. Playback may work ok, but recording seems to be a no-go.