TMRh20/TMRpcm #147: Play 2 simultaneous audio clips from SPI Flash (Winbond W25Q128)

eugjr, 2020-12-28:
Hi, first of all, great library you've created.
It is possible to store and acess the audio files from a SPI Flash Chip, like Winbond W25Q128, instead of SD Card?

I'm trying to develop a board to play audio clips for a model train, it's important to be able to play 2 simultaneous songs (engine song and horn), but the problem is, as the space is very narrow, the sd card adapter wont fit on the model train, resulting the need to use winbond chips.

Thanks

TMRh20, 2020-12-30:
Yeah it’s possible, but I have no experience with those chips. Just need to change the SD stuff to winbond code...

Theagainmen, 2021-05-11:
Any updates on this? I would like to play 2+ audio clips simultaneously too!

TMRh20, 2021-10-30:
ESP32 is not supported in TMRpcm. See the https://github.com/TMRh20/AutoAnalogAudio library

closed 2021-10-30 by TMRh20