TMRh20/TMRpcm #148: compilation error when using TMRpcm library for recording with sdfat library

zeeAkram, 2021-01-10:
Hi Sir.
I am trying to use TMRpcm library to record Audio with Sdfat library.
Installed TMRpcm and Sdfat libraries from following links:
https://codeload.github.com/TMRh20/TMRpcm/zip/master

https://codeload.github.com/greiman/SdFat/zip/master

then opened pcmConfig.h file in text editor from Documents\Arduino\libraries\TMRpcm-master, Uncommented following three lines:
#define SDFAT
#define ENABLE_RECORDING
#define BLOCK_COUNT 10000UL  // 10000 = 500MB   2000 = 100MB

first one was required for using SDFAT library with TMRpcm library, and other two were required for recording feature in TMRpcm library.

then open a built in example "sdfat" from TMRpcm >> xtraFeatures >> sdfat, and when tried to compile it following error occurred:
`Arduino: 1.9.0-beta (Windows 7), Board: "Arduino Uno"

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:195:2: error: 'Sd2Card' does not name a type; did you mean 'SdCard'?

  Sd2Card card1;

  ^~~~~~~

  SdCard

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp: In member function 'void TMRpcm::createWavTemplate(char*, unsigned int)':

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1609:5: error: 'SdVolume' was not declared in this scope

     SdVolume vol;

     ^~~~~~~~

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1609:5: note: suggested alternative: 'FsVolume'

     SdVolume vol;

     ^~~~~~~~

     FsVolume

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1616:7: error: 'card1' was not declared in this scope

  if (!card1.init(SPI_FULL_SPEED,CSPin)) {

       ^~~~~

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1620:6: error: 'vol' was not declared in this scope

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

      ^~~

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1620:6: note: suggested alternative: 'atol'

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

      ^~~

      atol

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1620:16: error: 'card1' was not declared in this scope

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

                ^~~~~

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1621:21: error: 'vol' was not declared in this scope

  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }

                     ^~~

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1621:21: note: suggested alternative: 'atol'

  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }

                     ^~~

                     atol

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1622:27: error: no matching function for call to 'SdFile::remove(SdFile*, char*&)'

  SdFile::remove(&rut, fNam);

                           ^

In file included from C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatVolume.h:28:0,

                 from C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatLib.h:27,

                 from C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/SdFat.h:34,

                 from C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:9:

C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatFile.h:718:8: note: candidate: bool FatFile::remove()

   bool remove();

        ^~~~~~

C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatFile.h:718:8: note:   candidate expects 0 arguments, 2 provided

C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatFile.h:733:8: note: candidate: bool FatFile::remove(const char*)

   bool remove(const char* path);

        ^~~~~~

C:\Users\CSharp\Documents\Arduino\libraries\SdFat-master\src/FatLib/FatFile.h:733:8: note:   candidate expects 1 argument, 2 provided

C:\Users\CSharp\Documents\Arduino\libraries\TMRpcm-master\TMRpcm.cpp:1631:9: error: 'card1' was not declared in this scope

    if (!card1.erase(bgnBlock, endBlock)){

         ^~~~~

exit status 1
Error compiling for board Arduino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

I tried my best to resolve this, worked many days and hours, searched on internet, tried different versions of both libraries and also tried with Arduino IDE version 1.8.11, but because of my limited knowledge in C++ , was not able to solve it, so decided to post here.

I hope you can help me. Thanks a lot.

Important Note: If i again comment following two lines of code , sdfat example compiles with out any error.
#define ENABLE_RECORDING
#define BLOCK_COUNT 10000UL  // 10000 = 500MB   2000 = 100MB

TMRh20, 2021-01-10:
Unfortunately recording isn’t currently working with the sdfat library so you would have to use the stock sd lib.

zeeAkram, 2021-01-10:
Thank you very much for your reply and this great library, hope you will be able to add this functionality in future.
one more thing: your note in example Recording sketch as in following link give a Feeling that Tmrpcm for Recording works with Sdfat .
https://github.com/TMRh20/TMRpcm/blob/master/examples/xtraFeatures/Recording/Recording.ino

closed 2021-01-10 by zeeAkram