TMRh20/TMRpcm #151: error when I try to record

rickyelqasem, 2021-04-05:
I get build errros listed below.. I uncommented the #define buffSize 128 AND #define ENABLE_RECORDING and the block part but when I build I get :

.pio\libdeps\nanoatmega328\PCM\TMRpcm.cpp: In member function 'void TMRpcm::createWavTemplate(char*, unsigned int)':
.pio\libdeps\nanoatmega328\PCM\TMRpcm.cpp:1609:5: error: 'SdVolume' was not declared in this scope
     SdVolume vol;
     ^~~~~~~~
.pio\libdeps\nanoatmega328\PCM\TMRpcm.cpp:1609:5: note: suggested alternative: 'volume'
     SdVolume vol;
     ^~~~~~~~
     volume
.pio\libdeps\nanoatmega328\PCM\TMRpcm.cpp:1616:13: error: 'class Sd2Card' has no member named 'init'
  if (!card1.init(SPI_FULL_SPEED,CSPin)) {
             ^~~~
.pio\libdeps\nanoatmega328\PCM\TMRpcm.cpp:1620:6: error: 'vol' was not declared in this scope
  if(!vol.init(&card1)){}//Serial.println("card failed"); }
      ^~~
.pio\libdeps\nanoatmega328\PCM\TMRpcm.cpp:1621:21: error: 'vol' was not declared in this scope
  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }
                     ^~~
*** [.pio\build\nanoatmega328\lib22f\PCM\TMRpcm.cpp.o] Error 1

TMRh20, 2021-04-05:
Recording with sdfat lib is broken so make sure to comment that out as well.

rickyelqasem, 2021-04-06:
OK . I will have to wait for it to be fixed. I cannot remove SDFAT from my project.. So for now no record feature.

rickyelqasem, 2021-08-20:
Has this been fixed as I see a new update?

TMRh20, 2021-08-20:
No still not fixed

TMRh20, 2021-10-30:
SdFat is fixed in latest release which should be avialable in 24hrs in Arduino library manager

closed 2021-10-30 by TMRh20