TMRh20/TMRpcm #152: Record Audio File Problem: Start Recording About 3 Seconds After the Command & Play Speed Too Fast & Cannot Be Opened on Computer

lindsay001, 2021-04-08:
Hi,

I am using TMRpcm library to record and play some wav files in my project. I'm using MEGA 2560, micro SD card and its adapter, max9814, pam8403(amplifier), and a 4 ohm speaker. 
I've tested the recording function before on my UNO board and it works totally fine even though there's some noise (the recording starts right away, the speed is correct, and the file can be opened on my computer). However, when I switch to MEGA board to attain more space for storage, I encountered some problems as indicated in the title: 
1. It starts recording with an approximate 3 seconds delay (I used the serial input as command); 
2. The file can be played on MEGA board, but the play speed is too fast. 
3. The saved wav file cannot be opened on my computer. 

I'm suspecting that something unexpected happens when saving the wav file on the micro SD card since the files already on the SD card can be played normally. But I cannot figure out what exactly happened and how to solve this problem... (Does it have something to do with the pcmConfig? But I used the same for UNO) 

I would really appreciate it if you can help me out! Below is the code:

`#include <SD.h>             
#include <TMRpcm.h>         
#include <SPI.h>
#define SD_ChipSelectPin 53  //example uses hardware SS pin 53 on Mega2560
TMRpcm tmrpcm;   

void setup(){
  tmrpcm.speakerPin = 11; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc
  Serial.begin(9600);
  if (!SD.begin(SD_ChipSelectPin)) {  // see if the card is present and can be initialized:
    Serial.println("SD fail");  
    return; 
  }
  else{   
    Serial.println("SD ok");   
  }
  tmrpcm.quality(1);
  tmrpcm.setVolume(5); 
}

void loop(){  
  if(Serial.available()){
    char temp = Serial.read();
    switch(int(temp)){
    case 'r': tmrpcm.startRecording("test.wav",16000,A4,0); Serial.println("start recording");break; // I found a delay after I send 'r'
    case 's': tmrpcm.stopRecording("test.wav"); Serial.println("stop recording"); break;
    case 'a': tmrpcm.play("1.wav"); break; //the other files are fine
    case 'b': tmrpcm.play("c1.wav"); break;
    case 'c': tmrpcm.play("test.wav"); break; // I use this one to play the test.wav
    case '?': if(tmrpcm.isPlaying()){ Serial.println("A wav file is being played");} break;
    case 'S': tmrpcm.stopPlayback(); break;
    default: break;
    }
  }
}`

lindsay001, 2021-04-08:
By the way, another issue I noticed is that when connecting the microphone OUT pin to some analog pins, I got a bigger noise and sometimes the files cannot be played (only noise on the speaker).

TMRh20, 2021-04-09:
First things first the recording feature is experimental so never worked 100% correctly. Next I would be looking at your power supply to see if there is enough power to handle writing to sd card at high speed. There isn’t much else that should be going wrong with it.

lindsay001, 2021-04-09:
Thank you for your reply! I think that's a good point. There should be no problem with the code and wiring, etc. So the wrong thing can only be the hardware I guess... 
I would like to also ask if you have suggestions for running a "complicated" program in which you need to read commands from serial; read, write txt file; and create, play wav file on SD card. That's the project I'm doing, and I found that even though I can record and play audio in the "simple" program I posted above, the wav files cannot be played in the "complicated" program. Do you also consider this a power supply problem? Or do you have any other thoughts?

TMRh20, 2021-04-10:
I’m not sure, it could have something to do with low memory, but that should not be such an issue on the mega. I really don’t have much to suggest. :(

closed 2024-01-30 by TMRh20