TMRh20/TMRpcm #155: Clicking/Beeping sound

Nonsensible, 2021-04-19:
So when it's just my speaker connected it makes a weird beeping sound and when i use my amp it makes a weird beeping sound. I have just been testing it with the basic example code. Am I supposed to not include the .wav in the file name or something?

Nonsensible, 2021-04-19:
Should I not include .wav in the file name or something?

HubbleCommand, 2021-05-28:
You do need the file extension.

What file are you trying to play ?
- It could be a problem with the date : make sure that it's sample rate is below 32kHz, and is 8 but unsigned.
- It could also be a problem with the name : make sure that the name is not more than 8 chars long (see https://github.com/TMRh20/TMRpcm/issues/117)

By basic example code you mean like https://github.com/TMRh20/TMRpcm/blob/master/examples/basic/basic.ino ? If not could you show your code ?

Nonsensible, 2021-06-15:
#include "SD.h"                      // need to include the SD library
#define SD_ChipSelectPin 4  //using digital pin 4 on arduino nano 328, can use other pins
#include "TMRpcm.h"          //  also need to include this library...
#include "SPI.h"

TMRpcm tmrpcm;   // create an object for use in this sketch

void setup(){
  tmrpcm.speakerPin = 9; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc

  Serial.begin(9600);
  if (!SD.begin(SD_ChipSelectPin)) {  // see if the card is present and can be initialized:
    Serial.println("SD fail");  
    return;   // don't do anything more if not
  }
  tmrpcm.setVolume(4); 
  tmrpcm.play("poweron.wav"); 
  delay(3000);
}

void loop(){ 
    
    tmrpcm.play("hswing1.wav");
    delay(5000);
    tmrpcm.play("swing1.wav");
  }
  

Right now I'm just testing sounds I have on the sd card. The beeping isn't a problem anymore, but every time a sound file starts, it makes a weird clicking sound. i converted all the files to the requirements

Nonsensible, 2021-06-15:
the clicking actually happens every second or so

HubbleCommand, 2021-06-15:
Could it be related to https://github.com/TMRh20/TMRpcm/issues/74?

Nonsensible, 2021-06-15:
Yes thank you it works great now!

closed 2021-06-15 by Nonsensible