TMRh20/TMRpcm #156: Using a DAC between the Arduino and the speaker

rajapaksha-rmpap, 2021-04-27:
Hello! I am new to electronics and know very little about it. For my school project, I have to simulate a simple voice recorder using Arduino Uno in Proteus 8 software, and I'm using the "TRMpcm" library for both 'recording' and 'playing back audio' parts. 

Here is my problem. 

We all know that the Ardunio Uno does not contain an inbuilt Digital to Analog Converter(DAC) just like all the other boards in the Arduino Family. And therefore, it can not give an analog signal as the output anyway. So, it has to either use Pulse Width Modulation(PWM) [as we use with "analogWrite();" function] or Serial Communication to transmit data to a peripheral device, in my case to the speaker. But, in the description of the TRMpcm library, it describes itself as an _'Arduino library for asynchronous playback of PCM/WAV files direct from SD card'_ , and therefore, I assume that TRMpcm library uses 'asynchronous serial communication' to pass the digital data to the speaker. Since this transmitted data is digital, it requires a DAC to play this data using a speaker. 

However, in all most all of the videos on YouTube, people directly connect the Speaker to the Arduino Uno board only through an amplifier, and they are not telling anything about using a DAC. I believe the reason for that is all the actual speakers come up with an inbuilt DAC with it. (I do not know whether this assumption is correct or not, and if there is another explanation for this, please let me know.) But unfortunately, the speaker module in Proteus software does not come with an inbuilt DAC (as per my knowledge). 

Therefore, now I have to connect an external DAC between the Arduino and the speaker  in my simulation along with a suitable amplifier. But I have no idea about what kind of a DAC is suitable to convert this **asynchronous serial bit-stream** into **an analog signal** and what are the parameters that I have to consider when choosing a suitable DAC. Please, can someone tell me, what kind of method will help me. And if you do not know exactly about implementing these in Proteus, you can also describe me what are the actual DACs used in physical speakers to achieve this convertion.

TMRh20, 2021-04-27:
TMRpcm doesn’t support DACS it is asynchronous PWM output, in that audio processing is handled separately from the main loop. One can simply connect a speaker and/or amp.

AdrexTheQuick, 2021-05-14:
So basically, Pulse Width Modulation (PWM) uses a digital output pin to _simulate_ analogue output. To understand what this library is doing, you first have to understand what a .wav file is. A .wav file is basically a series of numbers (typically 12-24 bit) that represent air pressure. To _record_ a .wav file, you sample the air pressure (using a microphone) a certain number of times per second (typically 48,000, or 48 KHz) and write that to a file. To _play_ a .wav file, you feed those numbers to a DAC, which translates them to voltage on a pin, which causes a solenoid inside a speaker to move to a certain position, which causes the air pressure around the solenoid to change in roughly the same way that the air pressure was recorded. The problem is this: a digital output pin can only change from high state to low state (say 0v to 3v). PWM solves this by rapidly oscillating the digital output pin from high state to low state such that the _average_ voltage on the pin is equal to what the _constant_ voltage a DAC would output for the samples duration (for 48,000KHz, each sample is about 20 microseconds long). Obviously, this results in some reduction in sound quality (particularly at high frequencies) but it is simpler in terms of hardware.

closed 2021-10-30 by TMRh20