TMRh20/TMRpcm #157: Trying to play sound, just get buzzing.

OlClown, 2021-06-16:
I have been attempting to troubleshoot this for a while with no luck.
I have looked at this: https://github.com/TMRh20/TMRpcm/issues/39  It was no help.
I believe the sound is playing because the buzzing lasts for the duration of the audio. 
My speaker is driven by a simple emitter follower setup and I am using a UNO.

#include <SPI.h>
#include <pcmConfig.h>//Have tried with #define DISABLE_SPEAKER2 commented and uncommented
#include <pcmRF.h>
#include <TMRpcm.h>
#include <SD.h>
#define CS 4
TMRpcm MySD;

//pin 13 to sck, 12 MISO, 11 MOSI, 10 speaker, 4 CS
void setup() {
  
  Serial.begin(9600);
  MySD.speakerPin=10;
  SD.begin(CS);
  if(!SD.begin(CS)){
     Serial.println("SD Fail");
     return;
  }
  MySD.setVolume(7);
  MySD.quality(1);
  Serial.println("start");//Lets me know when the song is supposed to start playing.
  MySD.play("Song.WAV");// Song is "texasradiofish_-_Slow_Cooker" from "audio_samples"
  Serial.println(MySD.isPlaying());//Lets me know if the song is playing
}

void loop() {
  delay(2000);//does not do anything
}

TMRh20, 2021-06-16:
My best guess is that it has to do with the audio being played. Are you able to try with a different audio sample?

OlClown, 2021-06-16:
This is my different sample, I am using this sample because my other ones were doing the same thing.  I found "audio_samples" under the "code" tab and thought I could at least get one of those to work.  Here is a list of some I tried to no avail, I didn't even have to resample these but it still does the same thing.
[SampleWav.zip](https://github.com/TMRh20/TMRpcm/files/6666386/SampleWav.zip)

TMRh20, 2021-06-17:
Hmm, unfortunately I’m not sure what to suggest in that case.

OlClown, 2021-06-17:
Welp, I got it to work, and really only by chance, I was tired of working on this project on my laptop so I switched to my desktop and had to download the library again, Voilà!! it worked!  Maybe a bad download or something? Not sure but it works.

closed 2021-06-17 by OlClown