TMRh20/TMRpcm #158: using all functions at once

MichalZima, 2021-06-19:
hello, I tried to use this library for making a walkie-talkie but I couldn't make it work completely. It's because the library functions can't work all in once. you can only use different parts and functions of the library, depending on what DEFINITIONS you comment/uncomment. that works okay. I managed to record, play and send audio. but only as separate tasks and only if I edit config with specific comments/uncomments. The problem is, when I uncomment all of #DEFINE lines, nothig works. I want device to be able to record audio from microphone to SD, send it through nrf module or recieve audio from nrf module, also save it to SD and play it with speaker.

thanks for your help.

TMRh20, 2021-06-19:
Yeah that won’t work. Too much stuff going on. You could look at the rf24audio library but that still won’t quite do everything you want.

MichalZima, 2021-06-19:
Okay ...I need to find different solution. Thanks for your reply.

closed 2021-10-30 by TMRh20