TMRh20/TMRpcm #161: SdFat example not compiling

drp0, 2021-09-27:
With #define SDFAT  uncommented in pcmConfig.h and the latest version of sdfat (version 2) installed the example will not compile. This is a pity as the library works well under the SD library.

The example points to a sdfat library reference: http://code.google.com/p/sdfatlib/ which redirects to https://github.com/greiman?tab=repositories.

Error Listing:
C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:195:2: error: 'Sd2Card' does not name a type; did you mean 'SdCard'?

  Sd2Card card1;
  ^~~~~~~
  SdCard

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp: In member function 'void TMRpcm::createWavTemplate(char*, unsigned int)':

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1613:5: error: 'SdVolume' was not declared in this scope

     SdVolume vol;
     ^~~~~~~~

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1613:5: note: suggested alternative: 'FsVolume'

     SdVolume vol;
     ^~~~~~~~

     FsVolume

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1620:7: error: 'card1' was not declared in this scope

  if (!card1.init(SPI_FULL_SPEED,CSPin)) {

       ^~~~~

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1624:6: error: 'vol' was not declared in this scope

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

      ^~~

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1624:6: note: suggested alternative: 'atol'

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

      ^~~

      atol

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1624:16: error: 'card1' was not declared in this scope

  if(!vol.init(&card1)){}//Serial.println("card failed"); }

                ^~~~~

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1625:21: error: 'vol' was not declared in this scope

  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }

                     ^~~

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1625:21: note: suggested alternative: 'atol'

  if (!rut.openRoot(&vol)) {}//Serial.println("openRoot failed"); }

                     ^~~

                     atol

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1626:27: error: no matching function for call to 'SdFile::remove(SdFile*, char*&)'

  SdFile::remove(&rut, fNam);

                           ^

In file included from C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/FatLib/FatVolume.h:28:0,

                 from C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/FatLib/FatLib.h:27,

                 from C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/SdFat.h:34,

                 from C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:9:

C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/FatLib/FatFile.h:758:8: note: candidate: bool FatFile::remove()

   bool remove();

        ^~~~~~

C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/FatLib/FatFile.h:758:8: note:   candidate expects 0 arguments, 2 provided

C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/FatLib/FatFile.h:773:8: note: candidate: bool FatFile::remove(const char*)

   bool remove(const char* path);

        ^~~~~~

C:\Users\Patterson\Documents\Arduino\libraries\SdFat\src/FatLib/FatFile.h:773:8: note:   candidate expects 1 argument, 2 provided

C:\Users\Patterson\Documents\Arduino\libraries\TMRpcm\TMRpcm.cpp:1635:9: error: 'card1' was not declared in this scope

    if (!card1.erase(bgnBlock, endBlock)){

         ^~~~~

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

TMRh20, 2021-09-27:
This is a known issue and probably not getting fixed anytime soon :/

drp0, 2021-09-28:
I tried swapping in old sdfat versions of the library from https://github.com/greiman/SdFat/releases.
Starting at version 1.0.0 from 26 Apr 2017.
Nothing compiled on the example.
Could it be that your sdfat tested library was pre version 1.00?
1.00 appears to be the first version released for the arduino library manager and is probably not the first sdfat arduino library.
Can you provide a link to the sdfat library that tmrpcm sdfat.ino was known to work with?
If not, perhaps sdfat.ino should be removed from the pool of examples.

TMRh20, 2021-09-28:
It was an older version hosted on google code or something if I remember. That code is gone afaik. I’ll see about updating the library.

TMRh20, 2021-10-30:
Fixed in latest release https://github.com/TMRh20/TMRpcm/releases/tag/v1.2.2
Should be available in 24hrs in library manager

drp0, 2021-11-25:
Much thanks for the work.
The new sdfat example points to [library](http://code.google.com/p/sdfatlib/ )
This offers a re-direct link to [https://github.com/greiman?tab=repositories](https://github.com/greiman?tab=repositories)
I installed the latest version (2) of  Bill Greiman's sdfat library.
The tmrpcm sdfat example then compiled and ran correctly.
It would be very helpful for other users, if the example pointed directly to SDFAT library correctly.

David

closed 2021-11-25 by drp0