TMRh20/TMRpcm #163: Adding support for the Arduino Uno WiFi Rev2

ndrb, 2022-01-10:
Hey everyone. I just want to start by saying thank you for this super cool library, it's one of a kind and really helps expand the capabilities of most Arduino boards!
I wanted to use it with an Arduino Uno WIFI Rev 2 to have an internet connection possible as well. I have not found a library that works with that board unfortunately and I do not think this library supports it either.
I am trying to upload it and most of the errors I see center around undeclared variables:

```
error: 'SPR0' was not declared in this scope

         SPCR &= ~((1 <<SPR1) | (1 << SPR0));
```

```
error: 'SPI2X' was not declared in this scope

         SPSR |= (1 << SPI2X);

```

```
error: 'TOIE1' was not declared in this scope

     *TIMSK[tt] &= ~( togByte | _BV(TOIE1) );
```

```
error: 'TOIE1' was not declared in this scope

                 *TIMSK[tt] &= ~( togByte | _BV(TOIE1) );
```

I think there could be others as well since the stack race is rather long.
I don't know a lot about Arduinos and even less about libraries. But thinking in the vain of missing/undeclared variables; would this possibly be an easy fix to get the library to support other boards?

If a certain variable such as 'TOIE1' is not declared then maybe there is a corresponding PIN or value on the Wi-Fi rev 2 that can be used?
If I could get some help or get pointed in the right direction I would love to give a crack at this and try to update the library with some extra mappings if it could help me and other users. I would just need guidance/direction since I do not know much about Arduinos. 

Thank you kindly for your time :)

TMRh20, 2022-01-10:
The Uno Wifi is based on AVR architecture, and it looks not so different from the 328 based architecture, so it may be possible to bring in support for this device, but it is not currently supported.

>If a certain variable such as 'TOIE1' is not declared then maybe there is a corresponding PIN or value on the Wi-Fi rev 2 that can be used?

Maybe, it depends how the timers and interrupts function on this device. I haven't looked at it closely, but if you look at the following lines, you may be able to change the pointers to point to the correct registers and get things going (mostly) that way:
https://github.com/TMRh20/TMRpcm/blob/master/TMRpcm.cpp#L103-L115

Pretty much all the info you need is in the datasheet http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega4808-4809-Data-Sheet-DS40002173A.pdf

TMRh20, 2022-01-11:
In taking a closer look, it looks like a bunch of stuff would have to be re-written for this device. 
There is just not enough similarities in the architecture to make it easy.

closed 2022-12-09 by TMRh20