TMRh20/TMRpcm #164: One WAV file out of four will not play

rhaas79, 2022-02-10:
I have a project that I am working on with several high school students and came across your work. It is phenomenal and does EXACLTY what we were looking for! Thank you so much! I just have one issue with a single WAV file not playing.

Our Arduino project has four buttons which call the same function and pass a song name as a variable. Once called, the function stops any playback, moves a servo to a particular position (145 degrees), starts musing playback and cycles an air solenoid. Activity continues until isPlaying() evaluates to false. At that time the solenoid stops and the servo returns to its home position (30 degrees). Like I said, it's working very well....for three of our WAV files.

For the fourth file (attached as .zip because .wav is not supported) the music stops playing, the servo moves to 145 degrees, the speaker "pops" and the servo returns to 30 degrees. Music never plays and the solenoid never fires.

All four files were converted using Audacity with the same output parameters as described in your project page.
[Celebr8.zip](https://github.com/TMRh20/TMRpcm/files/8042018/Celebr8.zip)

Any assistance is appreciated.

Thank you.
.

TMRh20, 2022-02-10:
No problem.
I don't think there is any problem with the wav file, it is the correct format, so the problem may have to do with your code.
What library are you using for the servo? TMRpcm typically has problems with the standard servo library, due to the use of timers for audio playback. 
I suspect that your code is either using an incompatible servo library, or something else is wrong in the code that causes playback to fail.

rhaas79, 2022-02-11:
@TMRh20 thank you for your quick reply. I am using the standard servo library. Of the four songs, "song 4" is consistently the only song which won't play; regardless of the order in which buttons are pressed. That's the only reason I though the song was suspect. I am using the standard servo library. Is there a different library I should be using? Is there a way to force TMRPCM to use timer2 and keep timer1 for the servo? Should I place a delay (100 mS?) between servo movement and song playback?

TMRh20, 2022-02-11:
That seems a bit odd. Is there anything in your code that is done differently when song 4 is played? Can you post your code?
You can force TMRpcm to use timer2 by editing pcmConfig.h in the library directory and uncommenting #define USE_TIMER2 , but I don't know if that will work.

There is also a [Servo Timer 2](https://github.com/nabontra/ServoTimer2) library available that I think works with TMRpcm.
In the library manager, I also found SoftPWM which uses timer2 and could also be used to drive a servo.

rhaas79, 2022-02-11:
Nothing different happens with song four. Each button calls the same function and passes the song name as a parameter. Function is shown below:

`void piko(char songName[]) {
  tmrpcm.stopPlayback();                            // stop playing
  _isPlaying = false;
  for (goToPos = newPos; goToPos <= finalPos; goToPos += 1) {
      flute.write(goToPos);
      delay(55);
  }
  flute.detach();
  delay(25);
  newPos = --goToPos;
  digitalWrite(mutePin,HIGH);                       // unmute audio output
  tmrpcm.play(songName);
  _isPlaying = true;
  startMillis = millis();
  currentMillis = millis();
  prevLEDmillis = 0;
  digitalWrite(cyl, HIGH);
  while (_isPlaying) {                              // cycle air solenoid HIGH and LOW to move butterfiles
    /*
     * Volume control
     */
    theVol = analogRead(volCtrl);
    theVol = map(theVol, 0, 1023, 0, 63);
    setvolume(theVol);
    if (currentMillis - prevLEDmillis >= _cylPeriod) {
      if (LEDstate == LOW) {
        digitalWrite(cyl, HIGH);
        LEDstate = HIGH;
        prevLEDmillis += _cylPeriod;
      }
      else {
        digitalWrite(cyl, LOW);
        LEDstate = LOW;
        prevLEDmillis += _cylPeriod;
      }
    }
    currentMillis = millis();
    if(tmrpcm.isPlaying()) {
      _isPlaying = true;
    }
    else {
      _isPlaying = false;
      tmrpcm.stopPlayback();                        // stop playing
      digitalWrite(mutePin,LOW);                    // mute audio output
    }
  }
  digitalWrite(cyl, LOW);
  delay(25);;
  flute.attach(servoPin);  
  delay(25);
  for (goToPos = newPos; goToPos >= initPos; goToPos -= 1) {
      flute.write(goToPos);
      delay(55);
  }
  newPos = --goToPos;
  prevLEDmillis = millis();
}`

rhaas79, 2022-02-11:
ServoTimer2 did not solve the problem with song four, and my servo also stopped responding to commands.
Forcing TMRpcm to use timer2 caused music output to stop.

TMRh20, 2022-02-11:
Using timer2 will use different pins, 3 and 11, so the speaker needs to be connected to pin 3.
The file you uploaded works fine for me in a basic example, so it either has to be a servo library conflict or something in your code. If you haven't yet, you could test your song4 in a basic example without the servo to make sure it isn't messed up or something.

rhaas79, 2022-02-14:
I have downloaded _and slightly modified_ the basic example to allow for my existing button presses as opposed to Serial.read(). Song four is still not playing, although songs one, two and three are. Any other ideas?

```
#include <SD.h>                             // need to include the SD library
#include <TMRpcm.h>                         // also need to include this library...
#include <SPI.h>                            // also need to include this library...
#define SD_ChipSelectPin 10

TMRpcm tmrpcm;                              // create an object for use in this sketch

#define song1 2                             // use digital 2 to play song 1
#define song2 3                             // use digital 3 to play song 2
#define song3 4                             // use digital 4 to play song 3
#define song4 5                             // use digital 5 to play song 4

char file_song1[] = "1_Beauty.wav";
char file_song2[] = "2_Yellow.wav";
char file_song3[] = "3_Year.wav";
char file_song4[] = "Celebr8.wav";
uint8_t debounce = 150;                     // milliseconds of pushbutton debounce

void setup() {
  pinMode(song1,INPUT_PULLUP);
  pinMode(song2,INPUT_PULLUP);
  pinMode(song3,INPUT_PULLUP);
  pinMode(song4,INPUT_PULLUP);
  pinMode(SD_ChipSelectPin,OUTPUT);
  tmrpcm.speakerPin = 9;
  tmrpcm.setVolume(5);                      // 0 to 7. Set volume level
  tmrpcm.quality(1);                        // Set 1 for 2x oversampling. Set 0 for normal

  Serial.begin(9600);
  if (!SD.begin(SD_ChipSelectPin)) {        // see if the card is present and can be initialized:
    Serial.println("SD fail");  
    return;                                 // don't do anything more if not
  }
}

void loop(){  
  /*
   * Song buttons
   */
  /* 'song1' button is pressed */
  if ( !digitalRead(song1) ) {
    delay(debounce);
    tmrpcm.play(file_song1);
  }

  /* 'song2' button is pressed */
  if ( !digitalRead(song2) ) {
    delay(debounce);
    tmrpcm.play(file_song2);
  }

  /* 'song3' button is pressed */
  if ( !digitalRead(song3) ) {
    delay(debounce);
    tmrpcm.play(file_song3);
  }

  /* 'song4' button is pressed */
  if ( !digitalRead(song4) ) {
    delay(debounce);
    tmrpcm.play(file_song4);
  }
}
```

TMRh20, 2022-02-15:
I can only guess it is corrupted on the SD card or something like that. You could try reformatting the SD card or put the same song on with a different filename. The file plays fine for me, so I have no other ideas...

rhaas79, 2022-02-24:
I am at a loss. I have tried:

- reformatting the card and rewriting songs
- formatting a new card and rewriting songs
- using a different song in place of "song4"

Nothing has helped. The only consistent failure is when I press the button for "song4". The servo moves, the speaker "clicks", then the servo moves back to it's starting position. It is always with anything I try to play with the song four button. Very odd...

TMRh20, 2022-02-26:
Have you tried using a different pin for button4? What about a button5? Lol I dunno, this is an odd one…

rhaas79, 2022-02-28:
I ended up re-downloading the problematic track and ran it through Audacity several times with varying frequencies and output file formats. For some reason not all frequencies between 16kHz and 32kHz worked for this particular file. 22050 Hz worked for this song, but other options failed to produce sound. This issue can now be closed. Thank you for all of you help, your willingness to support the library and your quick responses! I'll be using this at home for some Halloween props this year for sure!

closed 2022-02-28 by rhaas79