TMRh20/TMRpcm #170: Audio "popping" during stereo playback and Pass-through recording not working

drp0, 2022-06-11:
This library is quite an achievement in a limited cpu!
I am using a mega.
The following pcmConfig.h is in use:
#define buffSize 254
#define SD_FULLSPEED
#define SDFAT
#define STEREO_OR_16BIT
#define ENABLE_RECORDING
#define BLOCK_COUNT 10000UL

1) During playback of a 2 channel 8bit 16000 Hz wav file the left channel exhibits an audible level of hum/popping noise. This does not appear with mono wav files. Swapping the speakers moves the distortion to the right side. 
Both channel pins are passed to a small amplifier through 1uF capacitors.
Any hardware/software recommendations for aleviating this?

2) I can record wav files with satisfactory results. However I can not get pass-through to work at all. Any recommendations?

David

TMRh20, 2022-06-11:
Hmm, I’ll have to try recreating the issues here to give u a good answer I think. Gimme a bit of time…

TMRh20, 2022-06-12:
1. You might want to try using different timer pins. Mega supports pin pairs 5-2, 6-7,11-12,45-46. I'm guessing its due to some internal harmonics, so using a different timer may alleviate the issue. I haven't been able to recreate this issue.
2. Pass through wouldn't work for me using pin pairs 5-2 but worked using 11-12. Something odd going on there. I'm not sure of the cause yet, but at least there is a workaround.

drp0, 2022-06-12:
I was already working on pins 11 and 12 plus analogue 1.
As per your suggestion I tried 5-2 and 6-7.
I also tried Analogue 0.
Pins 45/46 not tested as they are covered by the sd card.
 No change in the left channel distortion.
No passthrough.
Just in case the amp was on the fritz I swapped the wires to 11/12 and the distortion moved to the right.
Recording with pass-through is recording.
Should passthrough work with stereo enabled? I did try with it off- still no pass-through.

How would you recommend connecting to an amp?
I tried 100nF, 1uF and 10 uF series capacitors- no difference to the distortion.

David

Sorush-moradisani, 2023-09-14:
I'm using Arduino mega with PAM8403 amplifier. the volume was fine but had tremendous amount of noise and squeaking. 
you can completely remove noise and relatively improve sound quality by connecting a 10uf capacitor between output pin and GND (or input of amplifier) .