TMRh20/TMRpcm #173: Stereo 2pins not working

MiguelCoelhoua, 2022-09-06:
I'm trying to play a stereo file, but only audio is coming out of one pin (pin 9).
When I pinMode(10,OUTPUT) the SDcard gives fail.
I'm using a shield for the arduino uno.

TMRh20, 2022-09-06:
I’m goin to guess that your shield uses pin 10 for the SD card so it’s likely you can’t use stereo mode with that shield

MiguelCoelhoua, 2022-09-06:
I read that in the shield datasheet it said this: "Arduino communicates with both the W5500 and the SD card using the SPI bus (through the ICSP header). This is on digital pins 10, 11, 12, and 13 on the Uno and pins 50, 51, and 52 on the Mega (...) and pin 4 for the SD card." Even if I can't use pin 10, I can't set another pin with outuput like for example 6 or 5. Like audio.speakerPin =9; pinMode(5,OUTPUT);

MiguelCoelhoua, 2022-09-11:
> I’m goin to guess that your shield uses pin 10 for the SD card so it’s likely you can’t use stereo mode with that shield
> […](#)

I read that in the shield datasheet it said this: "Arduino communicates with both the W5500 and the SD card using the SPI bus (through the ICSP header). This is on digital pins 10, 11, 12, and 13 on the Uno and pins 50, 51, and 52 on the Mega (...) and pin 4 for the SD card." Even if I can't use pin 10, I can't set another pin with outuput like for example 6 or 5. Like audio.speakerPin =9; pinMode(5,OUTPUT);

TMRh20, 2022-09-11:
The speaker pins are specific to the device, On the Mega, you can use 5-2,6-7,11-12,46-45 but on AVR devices, you can only use pins 9 and 10.

closed 2022-12-09 by TMRh20