TMRh20/TMRpcm #174: Anyway to increase the record volume?

rickyelqasem, 2022-09-10:
Been messing around with amps but I loose quality.  Any feature on the library to increase the saved wav?

TMRh20, 2022-09-10:
No feature built in no.

The best option would probably to increase the level of the input when recording. 

Alternately you could change the line https://github.com/TMRh20/TMRpcm/blob/master/TMRpcm.cpp#L658 to `buffer[whichBuff][buffCount] = ADCH << 1;` and change the 1 to a 2 or 3 maybe at the most. This doesn't typically give the best results compared to maximizing the input volume.

rickyelqasem, 2022-10-01:
OK bit more testing ... Since I'm trying to record a square wave anyway (data coming from old computers) this should work and it doesn't work well.  I'm wondering if you have the scope to add a "square mode" where it would record a pure square wave a bit like having a software Schmitt Trigger?

TMRh20, 2022-10-01:
Not really, the only thing I can think of would be if you match up the sample rate of the recording with the sample rate of the input for the best results. It may take some experimentation and tweeking though. Ie: 16025 or 15975 sample rate may match up with the computer better if it is outputting 16khz audio etc. because the AVR devices are not so precise with timing sometimes.

rickyelqasem, 2022-10-01:
worth a shot... I'll come back to you...

closed 2023-09-23 by TMRh20