TMRh20/TMRpcm #178: A Problem Playing Sound

OlClown, 2022-11-06:
I have successfully played sound with the entire setup I am playing with.  I know all the .wav are correct because I have played them before.  I recently picked up the project again after a hiatus and found that I could not play any of the sounds I could previously play.  I know that TMRpcm.begin() is returning true and it does think that it is playing something as I have some debug serial.print() stuff constantly updating whether it is playing or not and it is outputting as expected, however, all I hear is a low frequency "clicking" is the best way to describe it.  I used an o-scope to capture the output on speaker pin 9(I am playing on Arduino Uno) and this is what I am seeing.
[image]

Once I turn kill the sound (calling .disable()) the waveform goes flat as expected.  I took this reading across a 1k ohm, if I hook it up to my little op-amp I can hear it.  I am starting to suspect a bad board(I have been known to fry them) so I have ordered some more but in the case that the board is not to blame, what could be the culprit?  Let me know if you need more info.
(pin 13 to sck, pin 12 to miso, pin 11 to mosi, pin 4 to CS just to confirm my sd adapter setup and I am able to open the files on a computer and play them)

TMRh20, 2022-11-06:
Did you previously modify your pcmConfig.h file? It could just be an incorrect setting.

OlClown, 2022-11-06:
Yes, here are my non-default settings(the defines that I uncommented):HANDLE_TAGS, DISABLE_SPEAKER2, and rampMega.
Here is my setup
`   TMRpcm MySD;
    #define CS 4
    if (SD.begin(CS) == false)
        Serial.println("Failed");
    MySD.speakerPin = 9;
    MySD.setVolume(5);
    MySD.quality(1);`

OlClown, 2022-11-07:
Alright, it was in fact the board.  I was not highly convinced that it was the board because I have another board that I tested and it did about the same thing, however, with a brand new board it worked no problem... so I must have 2 fried boards. 
Thank you for the help and speedy response!

closed 2022-11-07 by OlClown