TMRh20/TMRpcm #179: Clash with Mozzi Library

Petros0000, 2022-12-07:
Hi,

I understand this is not an issue with the actual library, but thought I'd ask in case there is a solution.

I'm building a toy for my kids that among other things has  a theremin(using the Mozzi Library) and can play some wav sounds when keys are pressed(using the TMRpcm library).

Unfortunately when I compile the code I get the following error:

libraries/Mozzi-master/MozziGuts.cpp.o (symbol from plugin): In function `output_buffer':
(.text+0x0): multiple definition of `__vector_20'
libraries/TMRpcm/TMRpcm.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Olimexino-Nano.

I understand that both libraries use Timer1, hence the clash. Is there a way to bypass this problem? An important aspect is that the way the sketch is setup, it will be either the theremin or the wav samples, never together. 

If I enable Timer2 in pcmConfig.h I get the following error:

/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:138:36: error: 'TIMSK2' was not declared in this scope
         volatile byte *TIMSK[] = {&TIMSK2};
                                    ^~~~~~
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:138:36: note: suggested alternative: 'TIMSK'
         volatile byte *TIMSK[] = {&TIMSK2};
                                    ^~~~~~
                                    TIMSK
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:139:37: error: 'TCCR2A' was not declared in this scope
         volatile byte *TCCRnA[] = {&TCCR2A};
                                     ^~~~~~
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:139:37: note: suggested alternative: 'TCCRnA'
         volatile byte *TCCRnA[] = {&TCCR2A};
                                     ^~~~~~
                                     TCCRnA
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:140:37: error: 'TCCR2B' was not declared in this scope
         volatile byte *TCCRnB[] = {&TCCR2B};
                                     ^~~~~~
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:140:37: note: suggested alternative: 'TCCRnB'
         volatile byte *TCCRnB[] = {&TCCR2B};
                                     ^~~~~~
                                     TCCRnB
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:141:36: error: 'OCR2A' was not declared in this scope
         volatile byte *OCRnA[] = {&OCR2A};
                                    ^~~~~
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:141:36: note: suggested alternative: 'OCRnA'
         volatile byte *OCRnA[] = {&OCR2A};
                                    ^~~~~
                                    OCRnA
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:142:36: error: 'OCR2B' was not declared in this scope
         volatile byte *OCRnB[] = {&OCR2B};
                                    ^~~~~
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:142:36: note: suggested alternative: 'OCRnB'
         volatile byte *OCRnB[] = {&OCR2B};
                                    ^~~~~
                                    OCRnB
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:143:35: error: 'TCNT2' was not declared in this scope
         volatile byte *TCNT[] = {&TCNT2};
                                   ^~~~~
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:143:35: note: suggested alternative: 'TCNT'
         volatile byte *TCNT[] = {&TCNT2};
                                   ^~~~~
                                   TCNT
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/pgmspace.h:90,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:28,
                 from /Users/petros/Documents/Arduino/libraries/TMRpcm/pcmConfig.h:20,
                 from /Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:5:
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:145:34: error: 'OCIE2B' was not declared in this scope
         const byte togByte = _BV(OCIE2B);
                                  ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:145:34: note: suggested alternative: 'OCIE0B'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp: In member function 'void TMRpcm::timerSt()':
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:226:23: error: 'WGM21' was not declared in this scope
     *TCCRnA[tt] = _BV(WGM21) | _BV(WGM20) | _BV(COM2B1); //Fast PWM with 0xFF (255) top
                       ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:226:23: note: suggested alternative: 'WGM41'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:226:36: error: 'WGM20' was not declared in this scope
     *TCCRnA[tt] = _BV(WGM21) | _BV(WGM20) | _BV(COM2B1); //Fast PWM with 0xFF (255) top
                                    ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:226:36: note: suggested alternative: 'WGM40'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:226:49: error: 'COM2B1' was not declared in this scope
     *TCCRnA[tt] = _BV(WGM21) | _BV(WGM20) | _BV(COM2B1); //Fast PWM with 0xFF (255) top
                                                 ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:226:49: note: suggested alternative: 'COM1B1'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp: In member function 'void TMRpcm::pause()':
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:401:33: error: 'OCIE2B' was not declared in this scope
             *TIMSK[tt] |= ( _BV(OCIE2B) | _BV(TOIE1) );
                                 ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:401:33: note: suggested alternative: 'OCIE0B'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp: In member function 'void TMRpcm::play(char*, long unsigned int)':
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:500:29: error: 'CS20' was not declared in this scope
         *TCCRnB[tt] &= ~_BV(CS20);
                             ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:500:29: note: suggested alternative: 'CS00'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:501:28: error: 'CS21' was not declared in this scope
         *TCCRnB[tt] |= _BV(CS21);
                            ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:501:28: note: suggested alternative: 'CS01'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:503:29: error: 'CS21' was not declared in this scope
         *TCCRnB[tt] &= ~_BV(CS21);
                             ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:503:29: note: suggested alternative: 'CS01'
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:504:28: error: 'CS20' was not declared in this scope
         *TCCRnB[tt] |= _BV(CS20);
                            ^
/Users/petros/Documents/Arduino/libraries/TMRpcm/TMRpcm.cpp:504:28: note: suggested alternative: 'CS00'
exit status 1
Error compiling for board Olimexino-Nano.

Thanks for the help!

TMRh20, 2022-12-07:
This board is based on the 32u4 which has a 10-bit timer4 instead of 8-bit timer2, so one would have to modify the code to use timer4, probably no easy task for the average guy building toys for his kids. :p Would be challenging for me as well.

Petros0000, 2022-12-07:
Thought so:)

Thanks for the quick reply!

I have a pro mini that works with timer2, but I'm super low on memory. This might be a stupid question, but if I remove all references of other boards from your library, will I gain some?

TMRh20, 2022-12-08:
Not really, the defines only include code for a specific device. The best way to save memory is by using PROGMEM to store your file names etc. 
There is a program memory example included with the library, and if you install from zip file/source you can use the 'F" macro to do it ie: `tmrpcm.play(F("tune.wav"));`

closed 2023-02-18 by TMRh20