TMRh20/TMRpcm #184: recording using a Mega2560

rickyelqasem, 2023-02-20:
I have recorded stuff using a Nano328 but in an effort to increase the buffer I started looking at a Mega2560 .. When I record I just get a low-frequency rumble noise. It tired directly attached on both pins A5 and A7 .. Is Anything special I need to set up on a Mega 2560?

TMRh20, 2023-02-23:
Nope, configuration in TMRpcm should be about the same. Make sure you are using the correct SPI pins for the SD Card. They are different on the Mega.

rickyelqasem, 2023-02-23:
If I was using incorrect pins for spi I wouldn’t be able to record a file or browse the sd card but I can do both . However it’s just noise being recorded. I will review the circuit again.

rickyelqasem, 2023-02-24:
It is connected somehow to my PCB... closing issue

closed 2023-02-24 by rickyelqasem