TMRh20/TMRpcm #191: Popping/ high pitched hum

Fronk14, 2023-05-29:
So im using Arduino IDE, to have it play a sound with two buttons, but after it plays the setup sound and goes to the loop. if i press a button, it will still play the audio but hum and at the end of the audio it will pop the speaker. here is the code. 
`#include <SD.h>
#include <SPI.h>
#include <Adafruit_NeoPixel.h>
#include <TMRpcm.h>

#define SD_ChipSelectPin 10
#define BUTTON_PIN 3
#define BUTTON_PIN 2
#define PIN        6
#define NUMPIXELS 15

int lastButtonState = HIGH;
int lastButtonState2 = HIGH;Adafruit_NeoPixel pixels(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

TMRpcm tmrpcm;
unsigned long lastDebounceTime = 0;  // the last time the output pin was toggled
unsigned long debounceDelay = 0; 

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.print("Initializing SD card...");
  
  if (!SD.begin(10)) {
    Serial.println("initialization failed!");
    while (1);
  }
  Serial.println("initialization done.");
  tmrpcm.speakerPin = 9;
  
  pinMode(3,INPUT);
  pinMode(2,INPUT);
  
  pixels.begin();
  
  for(int i=0; i<NUMPIXELS; i++) {
    pixels.setPixelColor(i, pixels.Color(255, 0, 0));

    }
  pixels.show();
  tmrpcm.setVolume(5);
  tmrpcm.play("HF.wav");
  
   

}
void loop() {
  // put your main code here, to run repeatedly:
  int reading = digitalRead(3);
  int reading2 = digitalRead(2);

  if (reading != lastButtonState) {
    // reset the debouncing timer
    if (reading) {
      Serial.println("Button is pressed");
      tmrpcm.play("Blue.wav");
      for(int i=0; i<NUMPIXELS; i++) {
        pixels.setPixelColor(i, pixels.Color(0, 0, 225));
        pixels.setBrightness(225);
      }
    }
      
  }
  else if (reading2 != lastButtonState2) {
    // reset the debouncing timer
    if (reading2) {
      Serial.println("Button2 is pressed");
      tmrpcm.play("Orange.wav");
      for(int i=0; i<NUMPIXELS; i++) {
        pixels.setPixelColor(i, pixels.Color(225, 10, 0));
        pixels.setBrightness(225);
      }
    }
      
  }
  pixels.show();

  lastButtonState = reading;
  lastButtonState2 = reading2;
  delay(50);
}
`

TMRh20, 2023-05-29:
I haven’t tested this but am willing to bet it’s internal noise related to the neopixel library. Have you tested without toggling the LEDs? I’m not sure there is much that can prevent the issue other than not using the LEDs, but I will check it in more detail later.

Fronk14, 2023-05-29:
i did try without the LEDs but it doesn't do it with the setups audio file and it has LEDs as well

TMRh20, 2023-06-03:
Sorry for the delayed reply, but it took me a while to test this out. I wasn't able to recreate the issue. It works fine for me, but I was using pin 4 for CS.

I suspect there may be a problem with using pin 10 for Chip Select for the SD card. Have you tried changing it to pin 4 or some other digital pin? TMRpcm outputs on pins 9 and 10, so that may be the problem. I couldn't even get it working properly with pin10 as CS.

closed 2023-07-28 by TMRh20