TMRh20/TMRpcm #192: support for ATtiny 1-series

Reisen-Wandern-Tauchen, 2023-06-02:
Hello,

I was wondering if there was any chance to add support for the newer ATtiny chips.
Looking at the ATtiny1616 for example, these devices offer:

> ***– One 16-bit Timer/Counter type A (TCA) with a dedicated period register and three compare channels
> – Two 16-bit Timer/Counter type B (TCB) with input capture***
> – One 12-bit Timer/Counter type D (TCD) optimized for control applications
> – One 16-bit Real-Time Counter (RTC) running from an external crystal, external clock, or internal RC oscillator

So I guess there should be a possibility for this. But my hardware programming are quite limited, so I don't really know where to start.
Just compiling it with the "megaTinyCore" libraries (https://github.com/SpenceKonde/megaTinyCore) thows a bunch of errors related to the timer registers. As expected.

Is there any chance someone may take a look into adding support of these chips? (they are way cheaper than the ATmega series and quite powerful for their size, so it may be worth the effort)
I'd be happy to collaborate, but my knowlege in hardware low level programming is quite limited yet...

TMRh20, 2023-06-02:
It has similar timer functionality as other devices, but the API is different, so its not just a matter of changing register values or aliases etc. From a quick look at the datasheet, the whole thing would have to be re-written to support these devices, so one would probably be better off creating a separate library altogether.

I was in the same position as you when I wrote this library originally, its a matter of going through the datasheet and figuring out how to create a PWM signal and feed it PCM data each cycle.

Reisen-Wandern-Tauchen, 2023-06-02:
Thanks a lot for your fast reply!
OK, I see the problem. :-)
Well, in that case I'll think about some alternative approach for my project.

closed 2023-06-02 by Reisen-Wandern-Tauchen