TMRh20/TMRpcm #202: Error with sdfat and recording

predomaquilare, 2024-03-04:
hello, i was trying to use the lib with my arduino nano, sd module and mic module
i followed the instructions in wiki about recording and using the sdfat lib, like uncomment some lines in pcmconfig
for the functions startrecording and stoprecording it solves, but now im with problems with the sdfat lib
the error image is right below
[image]

by the way sorry for the bad english

TMRh20, 2024-03-04:
Can you try opening TMRpcm.cpp and edit this line https://github.com/TMRh20/TMRpcm/blob/master/TMRpcm.cpp#L1853
Change to `sFile.truncate(fPosition());`

closed 2024-03-04 by predomaquilare

TMRh20, 2024-03-04:
Thanks @predomaquilare for posting this issue, it has been fixed and a new release made which should be available in 24hrs.