TMRh20/TMRpcm #203: After calling analogRead(analogPin), the.wav file has no sound in it

MordexWasiu, 2024-03-04:
Hello to all the wonderful Creators of this library, sorry for my English but I use Google translator.

 I have some problem probably due to lack of sufficient knowledge.

 The idea of ​​my project is as follows...
 On Arduino Nano
 I wanted to create a voice recorder with a Vox function

 That is, record a Wave file with the sound taken from the microphone

 When there is silence, nothing happens - and when the microphone detects an increase in the sound level, recording begins and continues until the sound reaches the "zero" level again.
 If the sound on pin A0 reaches a lower level, i.e. similar to 0, the recording is stopped and the file session is closed and the file name is saved as file1.wav...

 The Wave file must be as long as the sound occurred
 if there is no sound, the file 1.wav is saved and if there is still a pause in the sound, nothing happens...
 If, after any long break, the sound on the AO pin microphone appears again, another 2.wav file with a higher number in the name is recorded and the cycle repeats

 This is my expected idea

 With this simplest example, everything works

 audio.startRecording(filename, sample_rate, mic_pin);

 delay(1000);

 audio.stopRecording(filename);

 The.wav file is saved and the Mono sound quality is sufficient for me and everything is OK...

 But when After calling the function

 audio.startRecording(filename, sample_rate, mic_pin);

 While recording, I'm trying to measure whether sound occurs because I need to know when the sound ends, so I need this measurement...

  ... analogRead(analogPin)

 Trying in every way
 Not only in the main void loop but even in any additional loop outside the main loop"
 This
 Once you have finished recording via
 audio.stopRecording(filename)

 The.wav file is saved as expected, but without sound, there is no sound in the file, only silence...
 Even though the file contains data and its size is comparable to the length of the recording, it is also visible on the SD card and can be easily played on another computer, the file opens in the player but there is no sound and silence...

 I just noticed that if between recording and stopping the recording there is an additional task in the form of analogread, even from a pin other than the A0 microphone, the .wav file is saved as a file without sound. If there is only a Delay or some other internal counter counting down the time before stopping the recording, everything is fine

 What can be done about it?  I will be grateful for any answer. Best regards!

TMRh20, 2024-03-04:
Instead of performing an analog read, you can read from the ADC register directly:

uint8_t soundLevel = ADCH;

This will give you a value between 0 and 255 that you can use to detect audio.

MordexWasiu, 2024-03-04:
> Instead of performing an analog read, you can read from the ADC register directly:
> 
> `uint8_t soundLevel = ADCH;`
> 
> This will give you a value between 0 and 255 that you can use to detect audio while recording, when recording is stopped you can use `analogRead()`.

Thank you very much for your answer, I will try to solve my problem thanks to your suggestion. If I succeed, I will be proud and thank you. Best regards👍🤜🤛

TMRh20, 2024-03-04:
Cool, also I think the following will give you the full resolution like `analogRead()` if that is needed
```
uint16_t soundLevel = ADCL >> 6;
soundLevel |= ADCH << 2;
```

MordexWasiu, 2024-03-04:
> Świetnie, myślę też, że poniższe informacje dadzą pełną rozdzielczość, `analogRead()`jeśli będzie to potrzebne
> 
> ```
> uint16_t soundLevel = ADCL >> 6;
> soundLevel |= ADCH << 2;
> ```

Dziękuję bardzo za odpowiedź, na razie muszę zapoznać się z operacjami bitowymi i bajtowymi, aby zrozumieć Twoją sugestię i podpowiedź.  Już nad tym pracuję z tym filmem... https://youtu.be/v0G-ddUeWpo?si=ZY2Wy2O71TYX4ndm
a wszystko dlatego, że szanuję Twój czas i nie chcę pytać o rzeczy, które są dla Ciebie oczywiste i dla mnie niekoniecznie... Twoja ostatnia odpowiedź ma wiele wspólnego  z tym co jest na filmie więc jestem bardzo szczęśliwy, że jestem na dobrej drodze chętnie poinformuję później czy udało mi się to osiągnąć i na razie jeszcze raz bardzo dziękuję😉👍

closed 2024-04-12 by TMRh20