TMRh20/TMRpcm #205: Crackling noise when trying to record

anonymouspenguin000, 2024-03-17:
I want to make an audio recorder as a project. So I did everything what is required in the [Audio Recorder documentation](https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features#recording-audio) and assembled my circuit. I use the Arduino Uno, the MAX4466 microphone module and a Class-10 MicroSD. My code is the built-in example of TMRPcm (xTraFeatures/Recording), the only thing I've changed is the CS pin to 10.

I run the code, open the Serial Monitor and see "SD OK". Then I do some recording (lowercase "r" to start, "s" to stop). There's actually a WAV file when I plug the SD into my PC, but it has strange loud crackling noise that makes hard to hear something real.

I thought that the problem can be because of Arduino Uno - it's a chinese Keyes Plus mod, so I tried to use the same code also on a (normal) Nano... And I got the same result. I also switched the mic, the same story. I tried to connect my Arduino to an external DC power supply, also the same story. It's been the 3rd day and I still get that noise.

Here's the archive with my WAV result, code and photos of my components - [Recorder.zip](https://github.com/TMRh20/TMRpcm/files/14627336/Recorder.zip)

TMRh20, 2024-03-17:
You are using the wrong pins. On arduino Uno the pin pairs are 9 and 10.

That being said you cannot use pin 10 for CS, well you might be able to if you disable it via pcmConfig.h #define DISABLE_SPEAKER2

anonymouspenguin000, 2024-03-17:
Thank you for a quick reply.
I did what you've said - DISABLE_SPEAKER2 in the config. And also:
```ino
audio.speakerPin = 9; //5,6,11 or 46 on Mega, 9 on Uno, Nano, etc
pinMode(SD_ChipSelectPin, OUTPUT);
// pinMode(12,OUTPUT);  //Pin pairs: 9,10 Mega: 5-2,6-7,11-12,46-45
```

...but unfortunately that didn't work :(
I also tried to switch the CS pin to 4 (did that in the circuit and programmatically), but I still get that noise

TMRh20, 2024-03-17:
Well there is something fundamentally wrong if its still doing that...

Have you
a: Tested the SD card & module with included SD sketches?
b: Tested the audio library with just playback?

TMRh20, 2024-03-22:
Sorry, I was set to reply to this issue, then promptly got distracted by other stuff.

Under examples in the Arduino IDE, you should find SD and some examples for the SD card like the ReadWrite example.

Are you able to test while powering the SD card from battery or external power supply? The only other thing I can think of is SD uses a lot more power to record than to read, because something is glitching out badly in any case.

You can also try using the SDFat library, there are instructions in the Wiki and you have to include and initialize the SDFat library and constructor per the SDFat example included with TMRpcm.

anonymouspenguin000, 2024-03-25:
Wow, that's it!! I've connected my SD card module to an external power supply and the problem has gone, now I get perfect recordings. Thank you!

TMRh20, 2024-03-25:
[image]

closed 2024-03-25 by TMRh20