TMRh20/TMRpcm #208: SD Card Read error while using TMRpcm

kausikp11, 2024-05-09:
I am working on a project where after recording the sound it needs to send the .wav file to local server. I tried working on sending the file seperately first which was working without issue.

When I tried integrating it with the TMRpcm it is showing as file not available. So when I checked it by unincluding the TMRpcm and its components, it works great.

Can you help me sort this as I need to record the sound and also send the file to server. The components I have are 

1) Arduino Nano.
2) Sound Sensor.
3) ESP-01 (Wifi module).

This is the code I am using

```
#include <TMRpcm.h>
#include "WiFiEsp.h"
#include <SD.h>
// Emulate EspSerial on pins 2/3 if not present
#ifndef HAVE_HWEspSerial
#include "SoftwareSerial.h"
SoftwareSerial EspSerial(7, 8); // RX, TX
#endif
//WiFiEspClient client;
WiFiEspClient client;
TMRpcm audio;
int file_number = 0;
static int len = 0;
#define CLIENT_BUFFER_SIZE 64 // Adjust this buffer size according to your needs

int buttonState = 0;
//1float flow; //Liters of passing water volume
//1unsigned long pulse_freq;

#define SD_CSPin 4
// delay function for with serial log.

void wait_min() {
  int count = 0;
  int secs = 5;
  while (1) {
    //Serial.print('.');
    delay(1000);
    count++;
    if (count == secs) {
      count = 0;
      break;
    }
  }
  //Serial.println();
  return ;
}
void setup() {
    // initialize serial for debugging
    SD.begin(SD_CSPin);
  Serial.begin(9600);
  //Sets up the pins
  pinMode(A0, INPUT);
  pinMode(3, OUTPUT);
  pinMode(6, INPUT_PULLUP);
//1  pinMode(2, INPUT);
//1  attachInterrupt(0, pulse, RISING); // Setup Interrupt

//  //Serial.println(F("loading... SD card"));
//  if (!SD.begin(SD_CSPin)) {
//    //Serial.println(F("An Error has occurred while mounting SD"));
//  }
//  else {
//    //Serial.println(F("SD card loaded"));
//  }
//  while (!SD.begin(SD_CSPin)) {
//    //Serial.print(F("."));
//    delay(500);
//  }
  
  // initialize serial for ESP module
  EspSerial.begin(9600);
  // initialize ESP module
  WiFi.init(&EspSerial);
  int status = WL_IDLE_STATUS;     // the Wifi radio's status

//  // attempt to connect to WiFi network
  while ( status != WL_CONNECTED) {
    // Connect to WPA/WPA2 network
    status = WiFi.begin("ACT103700825351", "25851250");
  }
  audio.CSPin = SD_CSPin;
}
void loop() {
  
  buttonState = digitalRead(6);
  char fileSlNum[3] = "";
    itoa(file_number, fileSlNum, 10);
    char file_name[10] = "";
    strcat(file_name, "spy");
    strcat(file_name, fileSlNum);
    strcat(file_name, ".wav");
  if (buttonState == LOW) {
//1    pulse_freq = 0;
//1    char dataString[20] = "";
    Serial.println("In loop");
    char fileSlNum[3] = "";
    itoa(file_number, fileSlNum, 10);
    char file_name[10] = "";
    strcat(file_name, "spy");
    strcat(file_name, fileSlNum);
    strcat(file_name, ".wav");
    //Serial.print("New File Name: ");
    //Serial.println(file_name);
    digitalWrite(3, HIGH);
    audio.startRecording(file_name, 84000, A0);
    //Serial.println("startRecording ");
    // record audio for 2mins. means , in this loop process record 2mins of audio.
    // if you need more time duration recording audio then
    // pass higher value into the wait_min(int mins) function.
    wait_min();
    audio.stopRecording(file_name);
    delay(1000);
    Serial.println("Out");
//1    flow = .00225 * pulse_freq;
    //Serial.println("stopRecording");
//1    Serial.println(flow);
//1    File dataFile = SD.open("dataset.csv", FILE_WRITE);
//1    strcat(dataString, file_name);
//1    strcat(dataString, ",");
//1    String data = dataString + String(flow);
//1    // if the file is available, write to it:
//1    if (dataFile) {
//1      dataFile.println(data);
//1      dataFile.close();
//1      // print to the serial port too:
//1      //Serial.println(dataString);
//1    }
//1    // if the file isn't open, pop up an error:
//1    else {
//1      //Serial.println("error opening datalog.txt");
//1    }
//1    pulse_freq = 0;
  //}
//  if (client.connect("192.168.0.104", 8000)) {
//    //Serial.println(F("Start uploading..."));
//  String start_request = "";
//  String end_request = "";
//  String filename="spy0.wav";
//  
//  start_request = start_request + "\n" + "------WebKitFormBoundaryuPbojkvQ2buPPB8a" + "\n" + "Content-Disposition: form-data; name=\"myfile\";filename="+filename+"\n" + "Content-Type: audio/wav" + "\n" + "\n";
//  end_request = end_request + "\n" + "-------WebKitFormBoundaryuPbojkvQ2buPPB8a--" + "\n";
//  len=len+start_request.length();
//  len=len+end_request.length();
//  client.println(F("POST /home HTTP/1.1"));
//  client.println(F("Host: 192.168.0.104"));
//  client.println(F("Content-Type: form-data; boundary=----WebKitFormBoundaryuPbojkvQ2buPPB8a"));
//  client.print(F("Content-Length: "));
//  client.println(String(len));
//  client.print(start_request);
//
//  File audio_file = SD.open("spy0.wav");
//  if(audio_file){
//    byte clientBuf[64];
//    int clientCount =0;
//    while(audio_file.available()){
//      clientBuf[clientCount] = audio_file.read();
//      clientCount++;
//      if(clientCount>63){
//        client.write(clientBuf,64);
//        clientCount=0;
//      }
//      if(clientCount>0) client.write(clientBuf,clientCount);
//      audio_file.close();
//    }
//    delay(1);
//    client.println(end_request);
//    client.stop();
//  }
//  
//  //Serial.println(end_request);
// }
sendFileToServer(file_name);
    digitalWrite(3, LOW);
    file_number++;
}
sendFileToServer("SPY0.WAV");
delay(30000);
file_number++;
}
void sendFileToServer(const char* fileName) {
  Serial.println(fileName);
//  String file;
//  file+=String(fileName);
//  Serial.println("In func");
//  Serial.println(file);
//  file.toUpperCase();
//  Serial.println(file);
  File audioFile = SD.open(fileName);
  if (!audioFile) {
    Serial.println("Error opening file");
    return;
  }

  // Connect to the server
  if (client.connect("192.168.0.104", 8000)) {
    Serial.println("Connected to server");

    // Send HTTP headers
    client.println(F("POST /simple_upload/ HTTP/1.1"));
    client.print(F("Host: "));
    client.println("192.168.0.104:8000");
    client.println(F("Connection: close"));
    client.println(F("Content-Type: multipart/form-data; boundary=----WebKitFormBoundary7MA4YWxkTrZu0gW"));
    client.print(F("Content-Disposition: form-data; name=\"myfile\"; filename=\""));
    client.print(fileName);
    client.println(F("\""));
    client.println(F("Content-Type: audio/wav")); // Set content type to audio/mpeg
    client.println(F("Content-Transfer-Encoding: binary"));
    client.print(F("Content-Length: "));
    client.println(audioFile.size());
    client.println();

//    Serial.println(F("POST /simple_upload/ HTTP/1.1"));
//    Serial.print(F("Host: "));
//    Serial.println("192.168.0.104");
//    //Serial.println(F("Connection: close"));
//    
//    Serial.print(F("Content-Disposition: form-data; name=\"myfile\"; filename=\""));
//    Serial.print(fileName);
//    Serial.println(F("\""));
//    Serial.println(F("Content-Type: audio/wav")); // Set content type to audio/mpeg
//    Serial.println(F("Content-Transfer-Encoding: binary"));
//    Serial.print(F("Content-Length: "));
//    Serial.println(audioFile.size());
//    Serial.println();
//    Serial.println(F("This is sending"));
//    Serial.println(audioFile.size());

    // Send the file content
    uint8_t buf[CLIENT_BUFFER_SIZE]; // Buffer to read from SD card
    size_t bytesRead;

    // Send the file content
    while ((bytesRead = audioFile.read(buf, sizeof(buf))) > 0) {
      client.write(buf, bytesRead);
    }

    // End of the request
    client.println();
    client.println(F("------WebKitFormBoundary7MA4YWxkTrZu0gW--"));

    audioFile.close();
    //Serial.println(F("File sent"));
  } else {
    Serial.println(F("Connection failed"));
    audioFile.close();
  }
}

//1void pulse () // Interrupt function
//1{
//1  pulse_freq++;
//1}
```

TMRh20, 2024-05-09:
I'm really not sure where the problem is, and I don't have the hardware to test, but I would guess the problem relates to the fact that the TMRpcm library kind of takes over the SD card as a global variable.

You can try re-initializing the SD Card AFTER a recording is complete: `SD.begin(SD_CSPin);` but I'm not sure if that will do the trick. 

The other option would be to include your sending code right into one of the TMRpcm functions like the `finalize wav template` function and access the SD card from within the library.

kausikp11, 2024-05-09:
I tried it but still getting the same issue, but when I put it seperately it works. Can you let me know that re initializing method or any other way it can be done.

I need this as this is the final part I am stuck with.

Attaching the code I used.

Arduino code:-
```
#include <TMRpcm.h>

TMRpcm audio;
int file_number = 0;
static int len = 0;
#define CLIENT_BUFFER_SIZE 256 // Adjust this buffer size according to your needs

int buttonState = 0;

#define SD_CSPin 4
// delay function for with serial log.

void wait_min() {
  int count = 0;
  int secs = 5;
  while (1) {
    //Serial.print('.');
    delay(1000);
    count++;
    if (count == secs) {
      count = 0;
      break;
    }
  }
  //Serial.println();
  return ;
}
void setup() {
    // initialize serial for debugging
  Serial.begin(9600);
  //Sets up the pins
  pinMode(A0, INPUT);
  pinMode(3, OUTPUT);
  pinMode(6, INPUT_PULLUP);

  int status = 0;     // the Wifi radio's status

  audio.CSPin = SD_CSPin;
  //audio.speakerPin = 3;
}
void loop() {
  
  buttonState = digitalRead(6);
  char fileSlNum[3] = "";
    itoa(file_number, fileSlNum, 10);
    char file_name[10] = "";
    strcat(file_name, "spy");
    strcat(file_name, fileSlNum);
    strcat(file_name, ".wav");
  if (buttonState == LOW) {

    Serial.println("In loop");
    char fileSlNum[3] = "";
    itoa(file_number, fileSlNum, 10);
    char file_name[10] = "";
    strcat(file_name, "spy");
    strcat(file_name, fileSlNum);
    strcat(file_name, ".wav");
    //Serial.print("New File Name: ");
    //Serial.println(file_name);
    digitalWrite(3, HIGH);
    audio.startRecording(file_name, 84000, A0);
    //Serial.println("startRecording ");
    // record audio for 2mins. means , in this loop process record 2mins of audio.
    // if you need more time duration recording audio then
    // pass higher value into the wait_min(int mins) function.
    wait_min();
    audio.stopRecording(file_name);
    delay(1000);
    Serial.println("Out");

    digitalWrite(3, LOW);
    file_number++;
}

file_number++;
}

```

TMRpcm modified code(added WiFiESP and updated the stopRecording):-
```
/*Library by TMRh20 2012-2014*/

#define RESOLUTION_BASE ((F_CPU) / 10)

#include <pcmConfig.h>
#if !defined (SDFAT)
    #include <SD.h>
#else
    #include <SdFat.h>
#endif
#include <TMRpcm.h>
#include "WiFiEsp.h"
#ifndef HAVE_HWEspSerial
    #include "SoftwareSerial.h"
    SoftwareSerial EspSerial(7, 8); // RX, TX
#endif

#if !defined (RF_ONLY)

void TMRpcm::stopRecording(const char *fileName){
    WiFiEspClient client;
EspSerial.begin(9600);
WiFi.init(&EspSerial);
while ( status != WL_CONNECTED) {
    // Connect to WPA/WPA2 network
    status = WiFi.begin("ACT103700825351", "25851250");
  }
    #define CLIENT_BUFFER_SIZE 256 // Adjust this buffer size according to your needs
    *TIMSK[tt] &= ~(_BV(OCIE1B) | _BV(OCIE1A));
    ADCSRA = adcsra;
    ADCSRB = adcsrb;
    ADMUX = admux;

    if(recording == 1 || recording == 2){
        recording = 0;
        #if defined (SDFAT)
            sFile.truncate(fPosition());
        #endif
        sFile.close();
        finalizeWavTemplate(fileName);
    }
    File sFile;
    sFile = SD.open(fileName);
  if (!sFile) {
    Serial.println("Error opening file");
  }

  // Connect to the server
  if (client.connect("192.168.0.104", 8000)) {
    Serial.println("Connected to server");

    // Send HTTP headers
    client.println(F("POST /simple_upload/ HTTP/1.1"));
    client.print(F("Host: "));
    client.println("192.168.0.104:8000");
    client.println(F("Connection: close"));
    client.println(F("Content-Type: multipart/form-data; boundary=----WebKitFormBoundary7MA4YWxkTrZu0gW"));
    client.print(F("Content-Disposition: form-data; name=\"myfile\"; filename=\""));
    client.print(fileName);
    client.println(F("\""));
    client.println(F("Content-Type: audio/wav")); // Set content type to audio/mpeg
    client.println(F("Content-Transfer-Encoding: binary"));
    client.print(F("Content-Length: "));
    client.println(sFile.size());
    client.println();

    // Send the file content
    uint8_t buf[CLIENT_BUFFER_SIZE]; // Buffer to read from SD card
    size_t bytesRead;

    // Send the file content
    while ((bytesRead = sFile.read(buf, sizeof(buf))) > 0) {
      client.write(buf, bytesRead);
    }

    // End of the request
    client.println();
    client.println(F("------WebKitFormBoundary7MA4YWxkTrZu0gW--"));

    sFile.close();
    //Serial.println(F("File sent"));
  } else {
    Serial.println(F("Connection failed"));
    sFile.close();
  }
}

#endif

#endif // Not defined RF_ONLY

```

kausikp11, 2024-05-09:
This is the wifi library I use Link:-https://github.com/bportaluri/WiFiEsp/tree/master

TMRh20, 2024-05-10:
I'm not sure I can be of much help on this, I typically only deal with library issues here, and this seems more like a coding issue. I would again suggest incorporating your SD sending code into the TMRpcm library, but there might be another way to go about this. You could also try the Arduino forums for more help.

closed 2024-07-17 by TMRh20