TMRh20/TMRpcm #212: Multitrack two pins example and Speed control

twilight102, 2024-06-06:
Good day,
Today I have an experiment day.
I would like to use audio speed control.
I do this with the Multitrack two pins example.
With these config settings:
#ifndef pcmConfig_h // if x.h hasn't been included yet...
#define pcmConfig_h // #define this so the compiler knows it has been included

#include <Arduino.h>
/****************** GENERAL USER DEFINES *************************** *****
#define buffSize 128 //must be an even number
#define SD_FULLSPEED
#define HANDLE_TAGS
//#define DISABLE_SPEAKER2
//#define USE_TIMER2
//#define debug
//#define SDFAT
#define ENABLE_MULTI
//#define STEREO_OR_16BIT
//#define MODE2 //Using separate 16-bit timers with up to 4 pins (Arduino Mega etc only)
#define disasterMega
//#define ENABLE_RECORDING
// #define BLOCK_COUNT 10000UL // 10000 = 500MB 2000 = 100MB
#define SPEED_CONTROL
//#define ENABLE_RF
//#define RF_ONLY

And the addition in the Multitrack two pins example, the setup line 36
 audio.setSpeed(50);
No matter what I try, I can't change the speed.
Can you please tell me what I am doing wrong?

Kind regards, Ebro de Wit

TMRh20, 2024-06-06:
Advanced Featuresgithub.comSee the wiki.You likely just need to tell it what timer to adjust.  tmrpcm.speedUp(10,0,1);Message ID: ***@***.***>

TMRh20, 2024-06-06:
Srry that post got messed up on my phone, see https://github.com/TMRh20/TMRpcm/wiki/Advanced-Features

Try calling
`tmrpcm.speedUp(10,0,1)` or `tmrpcm.speedUp(10,0,2) etc.`

closed 2024-07-17 by TMRh20