TMRh20/TMRpcm #214: Where can i find output pins by cpu type

pierrotm777, 2024-09-07:
Hello,

As i can see, Atmega32U4 is usable.
Where can i find which pin i can use as audio output ?
Is pin 6 usable ?
Add a table of audio output pins by cpu would be fine .

Thanks for your help.

TMRh20, 2024-09-07:
On Atmega32U4 the library would use Timer1, so it would be whichever pins are the output pins for Timer1.

This is defined in the datasheet and is different for each microcontroller, also different depending on what pins you use.

When using the Arduino platform, the pins are defined as pins 9 & 10 on Uno, Nano and similar boards based on the same MCU.

I'll think about adding more info to the Wiki, but users can contribute too.

pierrotm777, 2024-09-10:
Hello,
I have tried the pin 9 or 6 on my pro micro without success. On a pro mini, pin 9 run well.
Do I change something on the config file?
Thanks for your help.

TMRh20, 2024-09-11:
On Pro Mini the output pins are 9 and 10.

They should be the same on the Pro Micro, but I have never actually tested the library on a Pro Micro, as I don't have one.

pierrotm777, 2024-09-11:
Ok, je I shall try again.
Thanks

closed 2024-12-17 by TMRh20